DVMProject/dvmhost #6: Add Ability to have Redundant FNE address's

carcarjg, 2022-04-03:
Have the ability to have 2 fne address's configured. When it loses connection to the Pri it swaps to Secondary and checks for a good connection on Pri. I'm not asking this from safety of life viewpoint, but I don't want a wide area repeater to go down just because the auto-update went off/I tripped on the power cable of the main box/need to do FNE updates. I think it would be quite a nice feature.

lastation4, 2022-04-03:
I would suggest the architecture look something like this:

I'd put a Load Balancer in front of the two+ FNE instances and do it that way. Just my two cents.

carcarjg, 2022-04-03:
> 

I'd like it to be done in the Host. But for RN thats what I'm looking at doing. Just have to get my router to play nice with Kemp

gatekeep, 2022-04-03:
> I would suggest the architecture look something like this:
> 
> I'd put a Load Balancer in front of the two+ FNE instances and do it that way. Just my two cents.

I agree with the solution to place a proper load balancing configuration in front of the FNE instances. This would be the best and recommended approach, as it would provide a server-side load balancing/failover configuration, while requiring very minimal configuration on the host itself.

carcarjg, 2022-04-03:
> I agree with the solution to place a proper load balancing configuration in front of the FNE instances. This would be the best and recommended approach, as it would provide a server-side load [...]

Fair enough. My suggestion still stands as some of us are stuck with ISP routers and cant get the POS to detect kemps VIP XD

gatekeep, 2022-04-03:
> Fair enough. My suggestion still stands as some of us are stuck with ISP routers and cant get the POS to detect kemps VIP XD

Understandable. However, without coming off sounding like an asshole, the problem with the user network configuration using ISP routers and other networking technologies that may have their own difficulties while I agree, is annoying, is ultimately not our problem.

With that said; I will leave this ticket open with a low priority for now.

carcarjg, 2022-04-03:
> Understandable. However, without coming off sounding like an asshole, the problem with the user network configuration using ISP routers and other networking technologies that may have their own [...]

Oh yea, 100% understandable. Its more of a me problem than a you problem.