DVMProject/dvmhost #7: Duplex shift from iden_table not correct transmitted over the air

df1vb, 2022-04-15:
A duplex shift in the iden_table is not correct transmitted in the bandplan data over the air. The part dehind the decimal is cutted off.

Example iden_table entry:
0,438100000,6.25,-7.600,12.5,

Bandplan over the air:
Bandplan #0:   Base=438.1       Offset=-7      Spacing=6.25  BW=12.5

gatekeep, 2022-04-16:
Commit bdc9973bc5f71f6d291c42274c8edb5dc0f85c2a should fix this problem. There was a rounding issue due to the use of abs() vs fabs().

df1vb, 2022-04-17:
Sorry, but now we do have:

Bandplan over the air:
ID=0 Base=438.1 Offset=-7.59375 Spacing=6.25 BW=12.5

Example iden_table entry:
0,438100000,6.25,-7.600,12.5,

And in SN-DATA_CHN_ANN_EXP over the air we have:
DSO=00 TXch=438.175 RXch=438.175 DAC=0001

gatekeep, 2022-04-17:
Try the latest commit. I changed how this calculation was made, apparently the single line calculation and conversion from floating-point to integer caused weird rounding issues causing the issue. In my tests it seems to be properly calculating now.

df1vb, 2022-04-17:
Yepp, works fine!

closed 2022-04-17 by df1vb