DVMProject/dvmhost #8: SN-DATA_CHN_ANN_EXP  has wrong UPLINK frequency announced

df1vb, 2022-04-17:
SN-DATA_CHN_ANN_EXP  transmits wrong uplink frequency.

Block transmitted over the air: 
SN-DATA_CHN_ANN_EXP  DSO=00 TXch=438.175 RXch=438.175 DAC=0001

iden_table entrys:
# ChId,Base Freq,Spacing (khz),Input Offset (mhz),Bandwidth (khz),<newline>
0,438100000,6.25,-5.000,12.5,
1,145575000,6.25,-0.600,12.5,
2,438100000,6.25,-7.600,12.5,
3,439450000,6.25,-9.400,12.5,
4,438100000,6.25,-7.5875,12.5,

config.yml entrys:
    config:
        channelId: 0
        channelNo: c
        dmrNetId: 1
        voiceChNo:
            - c
        colorCode: 1
        nac: 13D
#         txNAC: 293
        pSuperGroup: FFFF
        netId: DEE00
        sysId: 13B
        rfssId: 1
        siteId: D

gatekeep, 2022-04-26:
this should have been fixed in ed4c7aca49b1b488c42031129bc266e51f805c90; can you pull and retry?

df1vb, 2022-04-27:
Its still messed up, now we have:

SN-DATA_CHN_ANN_EXP  DSO=00 TXch=438.2375 RXch=11-1960 DAC=0001
while:
RFSS_STS_BCST        SysID=13B LRA=0 RFSS=1 Site=13 CC=438.175 SC=70

gatekeep, 2022-04-28:
For now I've reverted the change in ed4c7ac. What application are you using to decode the CC data stream?

gatekeep, 2022-04-29:
Can you supply some examples of a decode using a similar IDEN_UP offset?

closed 2022-05-16 by df1vb