DVMProject/dvmhost #9: Rolling CC

carcarjg, 2022-04-18:
Add the ability to have mode change from CC to VC at a set interval. Would need to be coordinated with FNE?

Could also put in HS number and have it change daily with all HSs waiting until an internal counter rolls to them, Would need for all HSs to start at the same time

Or have the HS's assigned days of the week (HS1 = M,W,F HS2 = T,T,S HS3 = Sunday)

I think the last option would be the best unless we get the fabled site controller software

carcarjg, 2022-04-18:
Also added to FNE

gatekeep, 2022-04-18:
This would be a feature implemented at the soon-to-exist STC (Site Trunk Controller) FNE application, and not at the host.

closed 2022-04-18 by gatekeep

carcarjg, 2022-04-18:
> This would be a feature implemented at the soon-to-exist STC (Site Trunk Controller) FNE application, and not at the host.

Epic