DVMProject/dvmhost #37: Talkgroups With No Affiliations Are Repeated Reguardless Of Authoritative Or Supervisor Status

faultywarrior, 2023-07-24:
Title says it more or less entirely.  I've got a 2 channel (1 CC, 1 VC) site.  CC has Authoritative and Supervisor both set to true, VC has Authoritative set to true and Supervisor set to false.  Talkgroup on the FNE has Affiliated set to true.  I have only 1 radio powered on and affiliated to TG 31611.  My VC repeats traffic for all TGs regardless of an affiliation to it.  In my specific instance, TG 1056 happened to be very active; and was basically making the site unusable on any other TG.

FNE and DVMHost builds:
Host: https://github.com/DVMProject/dvmhost/commit/952606b3042527c467b6f9bab7ba710ea569f19c
FNE: https://github.com/DVMProject/dvmhost/commit/105811026383fa6ead10a6e13c4acb72e6c4a565

gatekeep, 2023-07-24:
If the intended behavior here is for the VC to only repeat the traffic commanded to by the CC, the VC should be set to authoritative false. A VC with authoritative true will repeat traffic of its own accord because it is authoritative in to do so.

Affiliation based repeat handling is done by the FNE, the FNE should be maintaining an internal table to validate whether or not a particular site has any affiliations to a TG before repeating the traffic to that site for an affiliation enabled TG.

faultywarrior, 2023-07-24:
This is correct.  The intended behavior is to only repeat traffic when commanded to.  Let me try setting my VC's authoritative false and see if that fixes it.  If not, then I suspect the issue is with the FNE's affiliation-based repeat logic not working as intended.

faultywarrior, 2023-07-24:
Setting my VC's authoritative false resolved the issue.

closed 2023-07-24 by faultywarrior