DVMProject/dvmhost #46: DMR Protocol

alwinhb, 2024-01-22:
@gatekeep @jelimoore @ratmandu @majmongoose @k4yt3x Does this repo contains source code for DMR data link and control layers? [Tier 1/2]

gatekeep, 2024-01-22:
Tier 1 is DMR simplex (I'm sure you meant Tier 2 / 3).

This depends on what you mean by DMR data link. If by data link you mean, an IP bearing interface no, there's no IP bearing interface support right now (ETSI-102 361-3). If you by data link mean, in-band DMR PDU and short data bearing services, then yes the dvmhost project contains code to handle and repeat in-band PDU data.

Yes, dvmhost implements the Tier 3 control layer for trunked operations.

alwinhb, 2024-01-23:
@gatekeep I couldn't locate TDMA and 4FSK

gatekeep, 2024-01-23:
It sounds like you are looking for how raw modulation is done; that isn't done within dvmhost, dvmhost is processing Layer 2 and 3 of the digital protocols. For raw modulation (Layer 1 essentially) you want to look at the firmware code. DVM Firmware: https://github.com/DVMProject/dvmfirmware and DVM Firmware Hotspot: https://github.com/DVMProject/dvmfirmware-hs.

alwinhb, 2024-01-23:
@gatekeep thanks you for the quick follow up. I am working on a custom hardware further I am trying to  implement dmr (tier 1/2/3) on that hardware. Ill look towards this repository for help!

closed 2024-01-24 by gatekeep