DVMProject/dvmhost #53: Conventional mode: Motorola GPS Location packets are not passed

jsb5151, 2024-04-22:
Currently have a repeater & a hotspot as a testbed, both connected to the same CFNE.

jsb5151, 2024-04-22:
As discussed out-of-band, this wouldn't need to be implemented; all it needs to do is pass it transparently for it to work (that packet goes through fine on non-Motorola P25 repeaters IIRC)

gatekeep, 2024-05-24:
Please try commit https://github.com/DVMProject/dvmhost/commit/5650db165719706c3e6f004497a16b28370181ec if this resolves the problem we can close this.

W3AXL, 2024-06-25:
DFSI needs some extra help, looks like the extra data in the MFID $90 packets is throwing off the src/dst ID decode, and possibly more:

```I: 2024-06-24 20:59:18.208 (SERIAL) V24 CALL START [STREAM ID 2715962297]
D: 2024-06-24 20:59:18.250 (SERIAL) V24 VHDR1 [STREAM ID 2715962297]
D: 2024-06-24 20:59:18.278 (SERIAL) V24 VHDR2 [STREAM ID 2715962297]
D: 2024-06-24 20:59:18.278 (SERIAL) P25, HDU algId = $80, kId = $0000, dstId = $0002
I: 2024-06-24 20:59:18.457 (SERIAL) P25, LDU1 (Logical Link Data Unit 1) audio, mfId = $00, srcId = 3118336, dstId = 2, group = 1, emerg = 0, encrypt = 0, prio = 4, lsd = 8288
I: 2024-06-24 20:59:18.638 (SERIAL) P25, LDU2 (Logical Link Data Unit 2) audio, algo = $80, kid = $0000, lsd = F170
D: 2024-06-24 20:59:18.815 (SERIAL) Got non-standard LC MFID: 90
M: 2024-06-24 20:59:18.815 (DUMP) RS-decoded LC (len 18)
M: 2024-06-24 20:59:18.815 (DUMP) 0000:  06 90 01 00 AC 3C C2 BF E4 00 00 00 00 00 00 00    *.....<..........*
M: 2024-06-24 20:59:18.815 (DUMP) 0010:  00 00                                              *..*
I: 2024-06-24 20:59:18.816 (SERIAL) P25, LDU1 (Logical Link Data Unit 1) audio, mfId = $90, srcId = 12763108, dstId = 3714108, group = 1, emerg = 0, encrypt = 0, prio = 1, lsd = AAAA
I: 2024-06-24 20:59:18.995 (SERIAL) P25, LDU2 (Logical Link Data Unit 2) audio, algo = $80, kid = $0000, lsd = AA82
I: 2024-06-24 20:59:19.175 (SERIAL) P25, LDU1 (Logical Link Data Unit 1) audio, mfId = $00, srcId = 3118336, dstId = 2, group = 1, emerg = 0, encrypt = 0, prio = 4, lsd = 88F1
I: 2024-06-24 20:59:19.356 (SERIAL) P25, LDU2 (Logical Link Data Unit 2) audio, algo = $80, kid = $0000, lsd = 70AA
D: 2024-06-24 20:59:19.537 (SERIAL) Got non-standard LC MFID: 90
M: 2024-06-24 20:59:19.537 (DUMP) RS-decoded LC (len 18)
M: 2024-06-24 20:59:19.537 (DUMP) 0000:  06 90 01 00 AC 3C C2 BF E4 00 00 00 00 00 00 00    *.....<..........*
M: 2024-06-24 20:59:19.537 (DUMP) 0010:  00 00                                              *..*
I: 2024-06-24 20:59:19.538 (SERIAL) P25, LDU1 (Logical Link Data Unit 1) audio, mfId = $90, srcId = 12763108, dstId = 3714108, group = 1, emerg = 0, encrypt = 0, prio = 1, lsd = AAAA
I: 2024-06-24 20:59:19.717 (SERIAL) P25, LDU2 (Logical Link Data Unit 2) audio, algo = $80, kid = $0000, lsd = 8288
I: 2024-06-24 20:59:19.895 (SERIAL) P25, LDU1 (Logical Link Data Unit 1) audio, mfId = $00, srcId = 3118336, dstId = 2, group = 1, emerg = 0, encrypt = 0, prio = 4, lsd = F170
I: 2024-06-24 20:59:20.079 (SERIAL) P25, LDU2 (Logical Link Data Unit 2) audio, algo = $80, kid = $0000, lsd = AAAA
I: 2024-06-24 20:59:20.148 (SERIAL) V24 CALL END```

(Notice any MFID=$90 LDU1 has the wrong srcId/dstId, and possibly the wrong priority)

gatekeep, 2024-07-09:
Was this resolved?

W3AXL, 2024-07-09:
b343473a87c87a9b24f4d8e53237c9c140538aff fixes this (hopefully) in DFSI

jsb5151, 2024-07-09:
Will update DVMDFSI as well as the other infrastructure elements, and will run the test plan again.  IIRC the audio dropouts are no longer there, but I'm not sure if the data is reliably passed.  Thanks guys.

gatekeep, 2024-07-26:
Is this resolved? Can we close this?

jsb5151, 2024-08-23:
Works fine across multiple hosts (both V24 and standard modem).  We can close this, thanks.

closed 2024-08-23 by jsb5151