DVMProject/dvmhost #74: LC::encodeLC(), unknown LC value, mfId = $00, lco = $C8 and 08

tsawyer, 2024-11-10:
I see these errors on occasion. What are they and are they useful?
```
Nov 09 15:28:45 homie-dvm DVM[10466]: E: (P25) LC::encodeLC(), unknown LC value, mfId = $00, lco = $C8
Nov 09 15:28:45 homie-dvm DVM[10466]: E: (P25) LC::decodeLC(), unknown LC value, mfId = $00, lco = $08
```

gatekeep, 2024-11-10:
- $C8 is impossible as the LCO field size is only 6 bits in size (maximum value of $3F). So this is either the result of incorrectly filtered data being passed, or a bug not properly filtering and based on the next line I'm gonna assume its incorrectly filtered.
- $08 is not valid, not such LCO opcode is defined by the TIA-102 standard.

These are both probably due to a bad decode.

tsawyer, 2024-11-10:
These errors are from DVMhost with a DVMV24 board. It's probably not relevant, as we have seen these all along, but both host and v2 board are updated to the very latest as of this date.

I'm reminded of the issue where the communications quality between DVMhost/DVMV24 connected Quantars is less than the repeaters talkback with weak signals. Whitelighting captured packet dumps for Discord which produced a conversation about the Quantar V.24 port sending garbage and the need for filtering or not. It seems these bad opcodes could be related to that issue.

This begs the wider question which is what to do about v.24 port garbage?